Sensirion/arduino-i2c-scd4x #4: Error: Not enough data received

histefanhere, 2021-08-14:
Hello,

This is an awesome sensor that works perfectly with my SensorBridge and monitoring its measurements in ControlCenter. However, when I try to use this library it throws and error! The example script reads the serial number of the sensor just fine, but then repeats this constantly:

```
Error trying to execute readMeasurement(): Not enough data received
Error trying to execute readMeasurement(): Not enough data received
Error trying to execute readMeasurement(): Not enough data received
Error trying to execute readMeasurement(): Not enough data received
...
```

Is there something that I need to configure on the sensor that isn't included in the example sketch? Thanks.

rnestler, 2021-08-16:
Hi @histefanhere 

The example should be working. Can you paste the full output, especially if perhaps `startPeriodicMeasurement` fails?

histefanhere, 2021-08-16:
This is the full output:

```
Serial: 0x<the chip's serial number>
Error trying to execute readMeasurement(): Not enough data received
Error trying to execute readMeasurement(): Not enough data received
Error trying to execute readMeasurement(): Not enough data received
Error trying to execute readMeasurement(): Not enough data received
...
```

And this repeats indefinitely. I verified with ControlCenter that the serial number being read is indeed the correct one, so it does have communication with the sensor. It also reads all the data fine via ControlCenter, so it's not the sensor value gathering itself that has an issue (at least I think).

rnestler, 2021-08-16:
In the example there is https://github.com/Sensirion/arduino-i2c-scd4x/blob/master/examples/exampleUsage/exampleUsage.ino#L95
```
    Serial.println("Waiting for first measurement... (5 sec)");
```

Which isn't visible in your output. Are you running the latest code?

Also which Arduino are you using? The error indicates that the return value from `requestBytes` did not match the expected number of bytes (https://github.com/Sensirion/arduino-core/blob/39c59326d3a8bed893a75a762dad9b82c4310c5e/src/SensirionI2CCommunication.cpp#L101) so maybe it is something specific to the Arduino board you are using.

histefanhere, 2021-08-19:
Ahhh, I think you might be onto something! In my version of the library (downloaded from the Arduino IDE Library Manager), I _don't_ have that Serial print. There's also another difference between the example code from the manager:

[image]

The return variables are given only 2 bytes each, but if it's returning floats, [which on Arduinos are 4 bytes](https://www.arduino.cc/reference/en/language/variables/data-types/float/) (I'm using an Arduino Nano based on the ATmega328P), that very well might be the cause of the issue! Unfortunetly I don't have the sensor with me at this time and lockdown has started up again here in NZ, but as soon as I get an opportunity to try it out I'll clone the repo and use the absolute latest version of the example sketch and see how it goes

histefanhere, 2021-08-19:
Update: I was able to access the sensor and I tried the latest version of the library, but still no luck. The output of the example script is now:

```Serial: 0x<sensor serial number>
Waiting for first measurement... (5 sec)
Error trying to execute readMeasurement(): Not enough data received
Error trying to execute readMeasurement(): Not enough data received
Error trying to execute readMeasurement(): Not enough data received
Error trying to execute readMeasurement(): Not enough data received
...```

rnestler, 2021-08-25:
> The return variables are given only 2 bytes each

We changed the driver to convert to float directly in the `readMesurement` mehod. See https://github.com/Sensirion/arduino-i2c-scd4x/blob/f055511173928e380a2d2b722f93ac7ce8db544a/src/SensirionI2CScd4x.cpp#L102

> Ahhh, I think you might be onto something! In my version of the library (downloaded from the Arduino IDE Library Manager),

The version in the library manager is indeed out of date. The 0.3.0 release seems to be broken, since the git tag doesn't match the version in `library.properties` (our fault). But for some reasons the 0.3.1 release didn't get published yet.

histefanhere, 2021-08-25:
Yeah when I tested it in my previous comment, I cloned the library straight from master so was using the latest version.

rnestler, 2021-08-25:
So back to my original question: Which Arduino board are you using? On some Arduino boards there are issues with I2C communication.

Another thing could be to use https://github.com/Sensirion/arduino-i2c-scd4x/blob/f055511173928e380a2d2b722f93ac7ce8db544a/src/SensirionI2CScd4x.h#L283
to check if data is available.
Or a https://github.com/Sensirion/arduino-i2c-scd4x/blob/f055511173928e380a2d2b722f93ac7ce8db544a/src/SensirionI2CScd4x.h#L328

histefanhere, 2021-09-02:
Got it working! It's probably the supply voltage that tripped it up - before I had it running from the 3.3V of the Arduino (in hindsight, not a good idea). I changed it to the Arduino's 5V and it's measuring data successfully! Thank you so much for all the help with this, I hope I didn't waste too much of your time :-)

closed 2021-11-03 by Sensirion-Bot