Sensirion/arduino-i2c-scd4x #8: scd41 connected to Atmega168PA

slechtvalk200, 2021-12-26:
I am trying to use the example code on a longan I2C CAN BUS module. I use a adafruit friend as FTDI. 

When i run this code : (stripped to bare minimum) 

```
#include <Arduino.h>
#include <SensirionI2CScd4x.h>
#include <Wire.h>

SensirionI2CScd4x scd4x;

void setup() {
  Serial.begin(115200);
  while (!Serial) {
    delay(100);
  }
  Wire.begin();
  uint16_t error;
  Serial.println("1");
  scd4x.begin(Wire);
  error = scd4x.stopPeriodicMeasurement();
  Serial.println(error);
 }

void loop() {
}
```

I get no println at all. Even not the  Serial.println("1");  So there has to be something very wrong, when i comment line  "error = scd4x.stopPeriodicMeasurement();" . There is no problem at all. Then i have all println, so then my code run. 

in arduino-1.8.15\libraries\arduino-i2c-scd4x\src\SensirionI2CScd4x.cpp , when i comment out "error = SensirionI2CCommunication::sendFrame(SCD4X_I2C_ADDRESS, txFrame,*_i2cBus);"  Then it runs again. 

Any idea what the problem could be?

psachs, 2022-03-23:
Hi, could you resolve the issue?
Can you also confirm that you get meaningful values with the minimal code snippet

I don't have a Atmega168PA to test. With several of the official Arduino boards everything works fine.
Arduino doesn't provide very good error handling system. If something goes wrong in the code, you may or may not get a print output. Serial communication writes to a buffer which might not get flushed in case of a crash.

The drivers internally use TwoWire which might cause problems with your platform. Else the commands should be the same as in the code snippets.

psachs, 2022-10-14:
closed due to inactivity

closed 2022-10-14 by psachs