Sensirion/arduino-i2c-scd4x #10: Automatic measurements: only a few received, then only errors.

stolk, 2022-01-07:
Running the example code on the SCD41 devkit:
```
Serial: 0x53CABF073B9F
Waiting for first measurement... (5 sec)
Co2:498	Temperature:22.40	Humidity:26.96
Co2:600	Temperature:22.34	Humidity:27.22
Co2:678	Temperature:22.20	Humidity:27.46
Error trying to execute readMeasurement(): Not enough data received
Error trying to execute readMeasurement(): Not enough data received
Error trying to execute readMeasurement(): Not enough data received
Error trying to execute readMeasurement(): Not enough data received
Error trying to execute readMeasurement(): Not enough data received
Error trying to execute readMeasurement(): Not enough data received
Error trying to execute readMeasurement(): Not enough data received
```

MBjoern, 2022-01-10:
Hi @stolk 

Looks like the code worked initially and then it failed. Hence, i'd assume a hardware issue here. Do you have proper connections to the sensor? No power interruptions, enough amps, no wiggly cables etc?

stolk, 2022-01-10:
Thanks.... I suspect this was a power issue.

The Arduino Pro Micro clone (Keye Studio) is a 5V microcontroller according to the spec, but I measured 4V at VCC, which dips below that at times. (Powered via USB.)

I have replaced the microcontroller with a different brand, that shows a proper 5V, and since doing that, this issue has gone away.

stolk, 2022-01-14:
I am still having power issues.

When the unit is measuring, I see 48mA draw at VDD. Is that nominal?

I noticed that if I wire as follows:

USB +5V to RAW input of Arduino.
VCC of arduino to SCD41.

...then I see a lot of noise on the power.
I observed this as follows:
LEDs with resistors connected to GPIO pins.

As the SCD41 measures... the LEDs flicker. When the measurement ends, the flicker stops.

I tried an alternative wiring:
+5V to Arduino RAW.
+5V to SCD41 VDD pin.

Then I do not see the flicker of the LEDs.

I have 0.1uF between GND and VDD pins.

How should I wire up the power for SCD41? (I use the Sensiron dev board.)

psachs, 2022-01-17:
Hi @stolk 
The peak current @5V is at 137mA. The average during periodic measurement should be around 13mA. So it is not unusual if you see 48mA e.g. during startup. However during regular measurement the average should be lower.

Power supply should preferably be stable with output ripple of less than 30 mV.

Connection of Arduino to SCD41 should be as follows (e.g. USB powered Arduino Uno R3)
* red cable VDD to 5V on Arduino (labeled as 5V on the Arduino)
* black cable Ground to Ground on Arduino
* green cable to SDA on Arduino (depends on board)
* yellow cable to SCL on Arduino (depends on board)

With the development board you shouldn't need any special setup (e.g. capacity, or pull up resistors).
Please make sure your power supply supports peak current. Even though it shouldn't be a problem, I suggest you don't connect an additional LED  for now since it will draw additional power. Since the Arduino seems to have issues with the power supply (at least the flickering of the LED suggest that) I suggest you check the specifications of the Arduino board.

I currently do not own an Arduino Pro Micro and can't test on this. I successfully tested the driver with the following boards:
* Arduino Nano
* Arudino Mikro WiFi 1010
* Arduino Uno R3
* Arduino Atmega 2560
* ESP32 WROOM 32D

Please find attached a photo of my working setup with an Arduino Uno R3.

[image]

stolk, 2022-01-17:
@psachs Thank you. Can you tell me what Voltage you measure at your VCC?

In automatic measurement mode, do you see that voltage go down for one second, ever five seconds?

I tried 3 different microcontrollers, they all show a drop. My esp32c3 for example, shows 5.02V dropping to 4.84V, for 1s every 5s. I am trying to avoid that w capacitors, but no luck so far.

psachs, 2022-01-17:
@stolk I currently don't have a reliable way to measure voltage. I will try to do so next time I get a chance.

A capacitor shouldn't negatively influence your setup as it helps stabilize the power supply (if correctly setup). For my setup it was never necessary so I don't use any additional resistors or capacity.

I have an ESP32 Wroom32 D (see attached picture) which works reliably with 3.3V and 5V. Do you have problems as well with your ESP32 or is it just the voltage dropping that shows up that concerns you?

[image]

stolk, 2022-01-17:
The esp32c is reliable but drops more than the 30mV you quoted. I see a
180mV drop w that controller.

SeeliSoft-CH, 2022-02-04:
Hi.
I ordered a SCD41 devkit and was also getting the "Not enough data received" error sometimes.
On a breadboard I have connected the SCD41 via the dupont jumpers (plus the male-male adapter) and a BME280 (also via I2C) to an ESP8266 (NodeMCU or similar china build).
First I had the impression the problem happens when I e.g. remove the BME280 which causes trouble on I2C protocol.
Then I also had the issue sometimes when I just moved the whole thing a little bit on the side.

So I came to the conclusion that the dupont jumper connectivity is a little bit fragile and the problem always happens when one the connections is interrupted for a very short time. This can happen e.g. by moving around the breadboard to touching the cables a little bit. I can reproduce the issue.

Since I'm aware of that I just programmed a self-health mechanism, as in such error case a stopPeriodicMeasurement() and a following startPeriodicMeasurement() command (with 1000 ms delay between as per the data sheet) brings the sensor back.

Btw I was also thinking of the lack of power since both sensors perform a measure from time to time which could lead in a situation that both sensor have the peak at the same time. But I believe that is not an issue as it runs fine over hours and I even had the issue having WiFi disabled on the ESP8266 and it also works fine with Wifi enabled (which takes a lot of energy) - as long as I don't touch the cables :)

stolk, 2022-02-04:
@SeeliSoft-CH Thanks. Could you measure the voltage you give the sensor.

Do you see it dip with every measurement? And if so, by how many mV?

I want to now if my 180mV dip is abnormal, or others have it too.

thanks!

SeeliSoft-CH, 2022-02-04:
@stolk Unfortunately I don't have a very precise multimeter. Nevertheless I can see that every ~4 seconds the voltage drops from 3.32 to 3.24. that would not mean 180mV in my case, only about ~80mV. But a drop is here. (for that test I removed everything except SCD41)

psachs, 2022-06-28:
The sensor performs a measurement every 5 seconds. According to our product team it is to be expected that the power drops during measurement, since the hotplate draws more current.
It is also possible that the drop is more then 30mV.

closed 2022-06-28 by psachs