Sensirion/arduino-i2c-scd4x #9: Highly suspect values from single-shot measurements.

stolk, 2022-01-07:
I am getting highly suspect values from single shot measurements.
As low as 217 ppm, with wild swings to 700, indoors.
I don't think they are correct.

I feed ambient pressure (991 currently, in my location) and wait for the data ready signal.

But I am getting nonsense back.

Is my sensor broken? Is single-shot function broken?

```
...
SensirionI2CScd4x cdos;
...
static void setupCO2Sensor(void)
{
  cdos.begin(Wire);
}

static void calibrateCO2Sensor(uint16_t pressure)
{
  const uint16_t err = cdos.setAmbientPressure(pressure);
  assert(err == 0);
}
```

In the main loop:
```
  // Request a co2 measurement.
  uint16_t err = cdos.measureSingleShot();
  assert( err == 0 );

  // Get the barometric pressure.
  const uint16_t pre = updateBarometer();
  
  // Calibrate the co2 device.
  calibrateCO2Sensor(pre);

  for (int attempt=0; attempt<10; ++attempt)
  {
    uint16_t rdy = 0xffff;
    uint16_t err = cdos.getDataReadyStatus(rdy);
    assert(err==0);
    Serial.println(rdy);
    if ( (rdy & 0x7ff ) != 0 )
    {
      uint16_t co2;
      float temperature;
      float humidity;
      uint16_t err = cdos.readMeasurement(co2,temperature,humidity);
      assert(err==0);
      Serial.println("co2:");
      Serial.println(co2);
      Serial.println("temperature:");
      Serial.println(temperature);
      Serial.println("humidity:");
      Serial.println(humidity);
      update_status_lines( pre, co2 );
      break;
    }
    if ( attempt==9 )
    {
      Serial.println("co2 data never became ready.");
    }
    delay(1000);
  }
```

Note: for some single shot measurements, the data rdy signal is never shown.

MBjoern, 2022-01-10:
Hi @stolk 

Thank you for reaching out to us. Did you consider the details mentioned in the datasheet regarding single shot measurements?

Here's the section from https://www.sensirion.com/fileadmin/user_upload/customers/sensirion/Dokumente/9.5_CO2/Sensirion_CO2_Sensors_SCD4x_Datasheet.pdf
The relevant part is marked in bold, but please read the whole thing. 

3.10 Low power single shot (SCD41)
In addition to periodic measurement modes, the SCD41 features a single shot measurement mode, i.e. allows for on-demand
measurements.
The typical communication sequence is as follows:
1. The sensor is powered up.
2. The I2C master sends a single shot command and waits for the indicated max. command duration time.
3. The I2C master reads out data with the read measurement sequence (chapter 3.5.2).
4. Steps 2-3 are repeated as required by the application.
**To reduce noise levels, the I2C master can perform several single shot measurements in a row and average the CO2 output
values. After a power cycle, the initial two single shot readings should be discarded to maximize accuracy.** The idle current in
between measurements is 0.15 mA (typ.), respectively 0.2 mA (max.). The energy consumed per single shot typically is 243 mJ
(296 mJ max.).
As for the periodic measurement modes, the automatic self-calibration (ASC) is enabled per default in single shot operation.
The automatic self-calibration is optimized for single shot measurements performed every 5 minutes. Longer measurement
intervals will result in less frequent self-calibration sequences. Note that no self-calibration is issued if the sensor is power-cycled
between single shot measurements Please consult Chapter 3.7 for a detailed description of the automatic-self calibration and
the corresponding commands.

stolk, 2022-01-10:
I suspect an unstable power supply has caused this issue.

I have moved to automatic sample mode, and a different microcontroller, and the measurement values are more stable now.

closed 2022-01-10 by stolk