Sensirion/arduino-i2c-scd4x #15: NAK on tansmit.

stolk, 2022-03-08:
So, this device was working fine for a month.
But suddenly, today, it would only read 0 ppm, out of the blue.

I programmed the **exampleUsage** code, to check, and I get:
```
Error trying to execute stopPeriodicMeasurement(): Received NACK on transmit of address
Error trying to execute getSerialNumber(): Received NACK on transmit of address
Error trying to execute startPeriodicMeasurement(): Received NACK on transmit of address
Waiting for first measurement... (5 sec)
Error trying to execute readMeasurement(): Received NACK on transmit of address
Error trying to execute readMeasurement(): Received NACK on transmit of address
Error trying to execute readMeasurement(): Received NACK on transmit of address
Error trying to execute readMeasurement(): Received NACK on transmit of address
Error trying to execute readMeasurement(): Received NACK on transmit of address
Error trying to execute readMeasurement(): Received NACK on transmit of address
```

This is with v 0.3.1 of the library.

Also I did an I2C scan, and the device does show up.

stolk, 2022-03-18:
Using:
```
    uint16_t testResult=0;
    const int tested = scd4x.performSelfTest(testResult);
    Serial.println(testResult);
```
I get the result code **196**

What does that code mean?

psachs, 2022-03-23:
Hi @stolk 

This looks like multiple errors from the self test are returned. I have to ask the product group how to correctly interpret the result.
Since factory reset did not fix the issue it might be a problem with the sensor

stolk, 2022-03-23:
@psachs Thank you Pascal.

The worrying thing is that out of a batch of 6 or so, two of my devices started having this symptom, **after having worked just fine for days.** 

I really want to get to the root of this, so please let me know what the product group says.

stolk, 2022-03-23:
@psachs Pascal, no, that is not the case: I use the SCD41 sensirion development kit. It has the sensor mounted on a PCB, and only has a VDD,GND,SCL,SDA pin out. 

I cannot have made a wiring mistake for the sensor: the sensor was wired by Sensirion themselves!

The board gets plenty of power: 5V 500mA from USB.

psachs, 2022-03-24:
The development kits should be wired correctly. I have several of them running directly connected to ESP32 without any problem. Since the self test fails it looks like its a problem with the sensor itself.

psachs, 2023-04-12:
@stolk 

Not sure how else to contact you. I received your sample and I was testing it for a couple of days. So far I couldn't reproduce your issue with it using our latest library.

I think a problem with the older Version of the example was, that we have to check for data_ready before reading out sensor values, as else it can in some cases cause a problem with the ASIC causing a crash of the sensor that can only be resolved with a reset.
I was not aware of this issue until recently and as it only seems to happen in corner cases it makes sense to me that chances are increasing the longer the sensor is running.

I will do some further testing to ensure I can run the sensor reliably and send you a replacement within the next few days.

stolk, 2023-04-12:
> Not sure how else to contact you. I received your sample and I was testing it for a couple of days. So far I couldn't reproduce your issue with it using our latest library.

Thank you, @psachs 

interesting… yeah, it could be some timing issue, I guess. Having said that, replacing the sensor board with another unit, using the exact same code on my microcontroller, resolved it for me.

So the problem can vary between sensors. Some units would be more susceptible than others, somehow.

psachs, 2023-04-28:
Hi @stolk 

Sorry for the delay. I would like to send you a replacement for the sensor you sent me.
Our logistics team just contacted me, that they can not send the package without your phone number

Can you please contact me at pascal.sachs at sensirion.com and send me the missing information?

adambennette, 2024-01-22:
Platform:  Arduino NANO EVERY  Atmega4809.

Hello,  I have four Sensiron SCD40 modules exhibiting the same semi-failure.  They were bought over a period of time from separate suppliers.
They respond to Serial Number request and return ACK for all other documented commands but never respond with sample data.  Issuing scd4x.getDataReadyFlag is ACK'd but always returns zero.  I have used the provided sampleusage sketch from Sensiron.  
I have a logic analyser on the I2C bus and can see all the transactions (Zeroplus LAP-C with protocol analyser).

All these devices worked for a long time before spontaneously failing in this way.  I suspect power-down mid-transaction but have no proof this is the cause.

These devices, prior to total failure seemed to hang the I2C bus with SDA held permanently low from time to time.  My hardware platform includes a relay to cut I2C power at reset for 2.5s which appeared to cure that particular problem, however, with the power cut there is a remaining trickle of current via the pullups into SCL and SDA.  Has this fried the device?  (10k pullups 3.3V VDD- seems unlikely to deliver much current to provoke a severe latch-up).

The SCD40 is quite expensive and my project is eating them for breakfast.  Any help appreciated.

adambennette, 2024-01-22:
... also I have tried issuing the reinit and performFactoryReset commands, both of which are ACK'd but change nothing.

Artellos, 2024-12-21:
> Using:
> 
> ```
>     uint16_t testResult=0;
>     const int tested = scd4x.performSelfTest(testResult);
>     Serial.println(testResult);
> ```
> 
> I get the result code **196**
> 
> What does that code mean?

For future reference, as I ran into this myself as well, the "Serial.println(testResult)" should probably be "Serial.println(tested)" as that is the output of the self-test.

- Olrik

andytrich, 2024-12-24:
I have the same problem.  The sensors are operating in a humid environment.  Could that be the cause?

LeonieFierz, 2025-02-05:
@andytrich 
Do you still face the problem with SCD4x returning no measurements for CO2?
If so, what is your setup?

closed 2025-02-25 by LeonieFierz