Sensirion/arduino-i2c-scd4x #22: How to use multiples sensors in the same arduino ?

pedrorocha014, 2023-02-09:
Hi everyone, 

I would like to use more than one sensor in my arduino uno, how can I use I2C to do this ?

Code exemple:

```

//---------------------
// SENSOR CO2
#include <Arduino.h>
#include <SensirionI2CScd4x.h>
#include <Wire.h>
SensirionI2CScd4x scd4x;
SensirionI2CScd4x scd4x_2;
//---------------------

//---------------------
// CARTAO SD
#include <SPI.h>
#include <SD.h>
File myFile;
int pinoSS = 10;  // Pin 53 para Mega / Pin 10 para UNO e NANO
//---------------------

//---------------------
// GPS
#define GPS_RX 0
#define GPS_TX 1
#define GPS_Serial_Baud 9600
#include <SoftwareSerial.h>
#include <TinyGPS.h>
TinyGPS gps;
SoftwareSerial gpsSerial(GPS_RX, GPS_TX);
int ano;
byte mes, dia, hora, minuto, segundo, centesimo;
//---------------------

//---------------------
// RELE
// byte rele = 3;
//---------------------

void setup() {

  Serial.begin(GPS_Serial_Baud);

  //---------------------
  // GPS

  gpsSerial.begin(GPS_Serial_Baud);
  //---------------------

  //---------------------
  // CARTAO SD
  pinMode(pinoSS, OUTPUT);
  SD.begin();
  myFile = SD.open("dados.txt", FILE_WRITE);
  //---------------------

  // SENSOR CO2
  //---------------------
  Wire.begin();

  uint16_t error;
  scd4x.begin(Wire, 0x3C);
  scd4x_2.begin(Wire, 0x3D);
  error = scd4x.stopPeriodicMeasurement();
  error = scd4x.startPeriodicMeasurement();
  //---------------------

  // RELE
  //---------------------
  // pinMode(rele, OUTPUT);
  //---------------------
}

void loop() {

  // delay(5000);

  //---------------------
  // GPS
  bool newData = false;
  unsigned long chars;
  // For one second we parse GPS data and report some key values
  for (unsigned long start = millis(); millis() - start < 5000;) {
    while (gpsSerial.available()) {
      char c = gpsSerial.read();
      // Serial.write(c); //apague o comentario para mostrar os dados crus
      if (gps.encode(c))  // Atribui true para newData caso novos dados sejam recebidos
        newData = true;
    }
  }
  //---------------------

  //---------------------
  // GPS

  //---------------------
  if (newData) {
    float flat, flon;
    unsigned long age;
    gps.f_get_position(&flat, &flon, &age);
    Serial.print("LAT=");
    Serial.print(flat == TinyGPS::GPS_INVALID_F_ANGLE ? 0.0 : flat, 6);
    Serial.print(" LON=");
    Serial.print(flon == TinyGPS::GPS_INVALID_F_ANGLE ? 0.0 : flon, 6);
    Serial.print(" SAT=");
    Serial.print(gps.satellites() == TinyGPS::GPS_INVALID_SATELLITES ? 0 : gps.satellites());
    Serial.print(" PREC=");
    Serial.print(gps.hdop() == TinyGPS::GPS_INVALID_HDOP ? 0 : gps.hdop());
    Serial.println();
    int ano1;
    byte mes, dia, hora, minuto, segundo, centesimo;
    gps.crack_datetime(&ano1, &mes, &dia, &hora, &minuto, &segundo, &centesimo);
    Serial.println(ano1);
    Serial.println("---------------------------------------------------------");
    //---------------------

    //---------------------
    // SENSOR CO2
    uint16_t error;
    uint16_t co2;
    float temperature;
    float humidity;
    error = scd4x.readMeasurement(co2, temperature, humidity);
    // Serial.print("Co2:");
    // Serial.print(co2);
    // Serial.print("\t");
    // Serial.print("Temperature:");
    // Serial.print(temperature);
    // Serial.print("\t");
    // Serial.print("Humidity:");
    // Serial.println(humidity);
    //---------------------

    //---------------------
    // CARTAO SD
    myFile = SD.open("dados.txt", FILE_WRITE);
    //---------------------

    //---------------------
    // CARTAO SD - GPS

    gps.crack_datetime(&ano, &mes, &dia, &hora, &minuto, &segundo, &centesimo);
    myFile.print(flat == TinyGPS::GPS_INVALID_F_ANGLE ? 0.0 : flat, 6);  // Escreve no Arquivo
    myFile.print(";");
    myFile.print(flon == TinyGPS::GPS_INVALID_F_ANGLE ? 0.0 : flon, 6);  // Escreve no Arquivo
    myFile.print(";");
    myFile.print(ano);
    myFile.print(";");
    myFile.print(mes);
    myFile.print(";");
    myFile.print(dia);
    myFile.print(";");
    myFile.print(hora);
    myFile.print(";");
    myFile.print(minuto);
    myFile.print(";");
    myFile.print(segundo);
    myFile.print(";");

    //altitude
    float altitudeGPS;
    altitudeGPS = gps.f_altitude();

    //velocidade
    float velocidade;
    velocidade = gps.f_speed_kmph();  //km/h

    //sentito (em centesima de graus)
    unsigned long sentido;
    sentido = gps.course();
    sentido = float(sentido) / 100;

    //satelites e precisão
    unsigned short satelites;
    unsigned long precisao;
    satelites = gps.satellites();
    precisao = gps.hdop();

    myFile.print(altitudeGPS);  // Escreve no Arquivo
    myFile.print(";");
    myFile.print(velocidade);  // Escreve no Arquivo
    myFile.print(";");
    myFile.print(sentido);  // Escreve no Arquivo
    myFile.print(";");
    myFile.print(satelites);  // Escreve no Arquivo
    myFile.print(";");
    myFile.print(precisao);  // Escreve no Arquivo
    myFile.print(";");

    // CARTAO SD - SENSOR CO2
    myFile.print(co2);  // Escreve no Arquivo
    myFile.print(";");
    myFile.print(temperature);  // Escreve no Arquivo
    myFile.print(";");
    myFile.print(humidity);  // Escreve no Arquivo
    myFile.println(";");
    myFile.close();
    //---------------------
  }
}

```

The error:

```
/Users/joaojcsq/Desktop/Programa_AREJA/source_us_com_011/source_us_com_011.ino: In function 'void setup()':
/Users/joaojcsq/Desktop/Programa_AREJA/source_us_com_011/source_us_com_011.ino:62:25: error: no matching function for call to 'SensirionI2CScd4x::begin(TwoWire&, int)'
   scd4x.begin(Wire, 0x3C);
                         ^
In file included from /Users/joaojcsq/Desktop/Programa_AREJA/source_us_com_011/source_us_com_011.ino:5:0:
/Users/joaojcsq/Documents/Arduino/libraries/Sensirion_I2C_SCD4x/src/SensirionI2CScd4x.h:56:10: note: candidate: void SensirionI2CScd4x::begin(TwoWire&)
     void begin(TwoWire& i2cBus);
          ^~~
/Users/joaojcsq/Documents/Arduino/libraries/Sensirion_I2C_SCD4x/src/SensirionI2CScd4x.h:56:10: note:   candidate expects 1 argument, 2 provided
/Users/joaojcsq/Desktop/Programa_AREJA/source_us_com_011/source_us_com_011.ino:63:27: error: no matching function for call to 'SensirionI2CScd4x::begin(TwoWire&, int)'
   scd4x_2.begin(Wire, 0x3D);
                           ^
In file included from /Users/joaojcsq/Desktop/Programa_AREJA/source_us_com_011/source_us_com_011.ino:5:0:
/Users/joaojcsq/Documents/Arduino/libraries/Sensirion_I2C_SCD4x/src/SensirionI2CScd4x.h:56:10: note: candidate: void SensirionI2CScd4x::begin(TwoWire&)
     void begin(TwoWire& i2cBus);
          ^~~
/Users/joaojcsq/Documents/Arduino/libraries/Sensirion_I2C_SCD4x/src/SensirionI2CScd4x.h:56:10: note:   candidate expects 1 argument, 2 provided
Multiple libraries were found for "SensirionI2CScd4x.h"
  Used: /Users/joaojcsq/Documents/Arduino/libraries/Sensirion_I2C_SCD4x
  Not used: /Users/joaojcsq/Documents/Arduino/libraries/arduino-i2c-scd4x-master
exit status 1

Compilation error: no matching function for call to 'SensirionI2CScd4x::begin(TwoWire&, int)'
```

LeonieFierz, 2023-02-16:
Hi @pedrorocha014 

The I2C address of SCD4x is fixed to 0x62, so you cannot have two SCD4x on the same I2C bus. A solution would be to use an I2C multiplexer. With this you can connect the two SCD4x on their separate bus and switch which bus to use from your code.

Hope this helps.

qfisch, 2023-02-20:
I will close that one since it has been answered.
Feel free to re-open it if you have further questions

closed 2023-02-20 by qfisch