Sensirion/arduino-i2c-scd4x #23: Received NACK on transmit of address

AndreasLichtsinn, 2023-02-16:
I was trying exampleUsage for Arduino, but with no success: 
[image]

It's a Pimoroni SCD41, connected via StemmaQT/Qwiic connector to Adafruit QT-Py ESP32 S2.
I had this working before with Adafruit WipperSnapper, but since my WiFi is instable I'm looking for a standalone solution.

qfisch, 2023-02-17:
Hello Andreas,

Unfortunately we do not have you hardware around for testing. But my best guess is the following.
Wire.h is not configured properly by default for your board. The library (by arduino) probably doesn't know the port number to use for I2C. Hence the failing communication with the sensor. 

If you find an example script for your board that establish an I2C connection using Wire.h, you can reuse the configuration in the exampleUsage here.

I hope it helps,
Let us know if we can support you further.

AndreasLichtsinn, 2023-02-17:
Ok makes sense, the Stemma port is a secondary I2C port, I will try this route, thanks.

AndreasLichtsinn, 2023-02-17:
[image]
Perfect, added Wire.setPins(SDA1, SCL1), very nice.

closed 2023-02-17 by AndreasLichtsinn