Sensirion/arduino-i2c-scd4x #24: Release a new version from master

brentru, 2023-03-03:
@LeonieFierz Hi - could you please cut and release a new version of this library so that it can be picked up by the Arduino Library Manager? There are a number of things added to `master` such as a new example with [some functions currently unavailable](https://github.com/Sensirion/arduino-i2c-scd4x/commit/4ec07965f30f32f4320960a04aadb7a19e6499c7) for use unless the user has manually tracked this library.

brentru, 2023-03-03:
Pls. remember to bump https://github.com/Sensirion/arduino-i2c-scd4x/blob/master/library.properties#L2 as well when you cut a release :)

qfisch, 2023-03-06:
Dear Brent (@brentru),

Thank you for that, it seems the release got forgotten along the way.
It should now be available.

Cheers,
Quentin

closed 2023-03-06 by qfisch