Sensirion/arduino-i2c-scd4x #25: How to determine the model number

learning88, 2023-03-17:
hello everyone, Can you tell SCD41 or SCD40 by the serial number?

qfisch, 2023-03-17:
Hello @learning88 ,

As far as I know there is no public mapping between serial and model.
A simple solution to distinguish them is to  request a single shot measurement as it will only work on SCD41. SCD40 would return an error.

Hope that helps you
Q

learning88, 2023-03-18:
@qfisch 

Ok, thank you

learning88, 2023-03-25:
I used measureSingleShot,scd40 did not return an error @qfisch

learning88, 2023-03-27:
hi  @qfisch 
no,I using arduino Pro.
Is there any other difference?
I tried looking at the current. Doesn't seem to make a difference

qfisch, 2023-03-27:
Can you link me to this SCD4x module you are using ?

learning88, 2023-03-27:
I bought the chip directly and soldered it onto the pcb I drew.
/*
 * Copyright (c) 2021, Sensirion AG
 * All rights reserved.
 *
 * Redistribution and use in source and binary forms, with or without
 * modification, are permitted provided that the following conditions are met:
 *
 * * Redistributions of source code must retain the above copyright notice, this
 *   list of conditions and the following disclaimer.
 *
 * * Redistributions in binary form must reproduce the above copyright notice,
 *   this list of conditions and the following disclaimer in the documentation
 *   and/or other materials provided with the distribution.
 *
 * * Neither the name of Sensirion AG nor the names of its
 *   contributors may be used to endorse or promote products derived from
 *   this software without specific prior written permission.
 *
 * THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND CONTRIBUTORS "AS IS"
 * AND ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT LIMITED TO, THE
 * IMPLIED WARRANTIES OF MERCHANTABILITY AND FITNESS FOR A PARTICULAR PURPOSE
 * ARE DISCLAIMED. IN NO EVENT SHALL THE COPYRIGHT HOLDER OR CONTRIBUTORS BE
 * LIABLE FOR ANY DIRECT, INDIRECT, INCIDENTAL, SPECIAL, EXEMPLARY, OR
 * CONSEQUENTIAL DAMAGES (INCLUDING, BUT NOT LIMITED TO, PROCUREMENT OF
 * SUBSTITUTE GOODS OR SERVICES; LOSS OF USE, DATA, OR PROFITS; OR BUSINESS
 * INTERRUPTION) HOWEVER CAUSED AND ON ANY THEORY OF LIABILITY, WHETHER IN
 * CONTRACT, STRICT LIABILITY, OR TORT (INCLUDING NEGLIGENCE OR OTHERWISE)
 * ARISING IN ANY WAY OUT OF THE USE OF THIS SOFTWARE, EVEN IF ADVISED OF THE
 * POSSIBILITY OF SUCH DAMAGE.
 */

#include <Arduino.h>
#include <SensirionI2CScd4x.h>
#include <Wire.h>

SensirionI2CScd4x scd4x;

void printUint16Hex(uint16_t value) {
    Serial.print(value < 4096 ? "0" : "");
    Serial.print(value < 256 ? "0" : "");
    Serial.print(value < 16 ? "0" : "");
    Serial.print(value, HEX);
}

void printSerialNumber(uint16_t serial0, uint16_t serial1, uint16_t serial2) {
    Serial.print("Serial: 0x");
    printUint16Hex(serial0);
    printUint16Hex(serial1);
    printUint16Hex(serial2);
    Serial.println();
}

void setup() {

    Serial.begin(115200);
    while (!Serial) {
        delay(100);
    }

    Wire.begin();

    uint16_t error;
    char errorMessage[256];

    scd4x.begin(Wire);

    // stop potentially previously started measurement
    error = scd4x.stopPeriodicMeasurement();
    if (error) {
        Serial.print("Error trying to execute stopPeriodicMeasurement(): ");
        errorToString(error, errorMessage, 256);
        Serial.println(errorMessage);
    }

    uint16_t serial0;
    uint16_t serial1;
    uint16_t serial2;
    error = scd4x.getSerialNumber(serial0, serial1, serial2);
    if (error) {
        Serial.print("Error trying to execute getSerialNumber(): ");
        errorToString(error, errorMessage, 256);
        Serial.println(errorMessage);
    } else {
        printSerialNumber(serial0, serial1, serial2);
    }

    // Start Measurement
//    error = scd4x.startPeriodicMeasurement();
//    if (error) {
//        Serial.print("Error trying to execute startPeriodicMeasurement(): ");
//        errorToString(error, errorMessage, 256);
//        Serial.println(errorMessage);
//    }

    Serial.println("Waiting for first measurement... (5 sec)");
}

void loop() {
    uint16_t error;
    char errorMessage[256];
    error=scd4x.measureSingleShot();
     if (error) {
       Serial.print("Error trying to execute startPeriodicMeasurement(): ");
       errorToString(error, errorMessage, 256);
       Serial.println(errorMessage);
   }
    delay(5000);

    // Read Measurement
    uint16_t co2;
    float temperature;
    float humidity;
    error = scd4x.readMeasurement(co2, temperature, humidity);
    if (error) {
        Serial.print("Error trying to execute readMeasurement(): ");
        errorToString(error, errorMessage, 256);
        Serial.println(errorMessage);
    } else if (co2 == 0) {
        Serial.println("Invalid sample detected, skipping.");
    } else {
        Serial.print("Co2:");
        Serial.print(co2);
        Serial.print("\t");
        Serial.print("Temperature:");
        Serial.print(temperature);
        Serial.print("\t");
        Serial.print("Humidity:");
        Serial.println(humidity);
    }
}

learning88, 2023-03-27:
output:
Serial: 0x39920F073B11
Waiting for first measurement... (5 sec)
Co2:993	Temperature:16.45	Humidity:64.15
Co2:990	Temperature:16.81	Humidity:63.20
Co2:984	Temperature:17.09	Humidity:61.85
Co2:935	Temperature:17.28	Humidity:60.87
Co2:899	Temperature:17.42	Humidity:60.17
Co2:896	Temperature:17.54	Humidity:59.70

learning88, 2023-03-27:
Here's another one:
Serial: 0x46E017073BA7
Waiting for first measurement... (5 sec)
Co2:1445	Temperature:17.47	Humidity:62.29
Co2:1422	Temperature:17.82	Humidity:61.16
Co2:1365	Temperature:18.01	Humidity:60.37
Co2:1345	Temperature:18.20	Humidity:59.67

qfisch, 2023-03-27:
There is no way to know from the measurements unfortunately.
I am looking internally if there is an identification method that I would not be aware of (it does not look good so far).
We are aware of this issue with SCD4x sensors. Sensors will be engraved in the future to avoid this situation.
I will for sure come back to you if I find a solution.

learning88, 2023-03-27:
@qfisch

Ok, thank you.