Sensirion/arduino-i2c-scd4x #27: The scd41 sensor only detects upto 5000 ppm, how do i use it to expand its range upto 40,000 ppm?

charankamal20, 2023-05-14:


Rol-la, 2023-05-15:
@charankamal20

Hi I'm not sure if I got you question properly.
The scd41 is returning a 16-bit value that is interpreted as CO2 in PPM. This means that a value of 40'000 (that's the maximum value that is returned by the sensor according to its data sheet) fits perfectly within theses 16 bits.  
The larger values require less accuracy.

charankamal20, 2023-05-16:
Hi @Rol-la
I am currently working on a project where i use this sensor to get the readings and display those readings on a lcd display and also send them to a server. The sensor readings on the screen were less than 5000, where the actual readings were around 10000. Me and my team after reviewing the code came to a conclusion that the sensor might be less than 5000 ppm but it wasn't the case as mentioned in the official documentation. 
So I was looking to see if the sensor had to be configured some way to measure values greater than 5000. 
But after looking at the database online, **_the values were right. The sensor was giving values upto 40,000 ppm range._** There was some issue in how the data was being displayed on the screen. Still working on that tho.
Thanks for responding to my issue!