Sensirion/arduino-i2c-scd4x #28: Unable to read CO2 but temperature and humidty works fine

antonio-emilio, 2023-06-17:
Hi everyone!
I've bougth a SCD41 sensor from mouser and developed my own custom board with it. I originally had connected 3V3 on it, but since the readings of the CO2 didn't work, I change it to 5V, but still doesn't work.

I've been receiving "Invalid sample detected, skipping.". When I remove this validation, I can see that CO2 is zero, but temperature and humidty readings are fine. What could be?

[image]

antonio-emilio, 2023-06-17:
Also, performing the self test didn't returned anything. The returned value is 0.

qfisch, 2023-06-23:
Hello @antonio-emilio ,

Interesting issue you have here.
Did you modify the example script outside of removing the validation ?
Do you have any other sensor on your custom board ?

antonio-emilio, 2023-06-24:
> Hello @antonio-emilio ,
> 
> Interesting issue you have here. Did you modify the example script outside of removing the validation ? Do you have any other sensor on your custom board ?

Actually, I managed to make it work.... turns out that I didn't solder the sensor very great... Power issue :P
But thank you a lot!

closed 2023-06-24 by antonio-emilio