Sensirion/arduino-i2c-scd4x #30: Example needed to print some internal values

domotiquefacile, 2024-01-04:
Hello,

I would like to confirm that self-calibration is enabled/disabled as well as the ASC Initial Period and Standard Period.
I have tried a (naive) approach:
```
uint16_t ascEnabled;
scd4x.getAutomaticSelfCalibration(ascEnabled);
Serial.println(ascEnabled);

uint16_t ascInitialPeriod;	
scd4x.getAutomaticSelfCalibrationInitialPeriod(ascInitialPeriod);
Serial.println(ascInitialPeriod);
```

But obviously this is not correct as I am getting weird outputs.

Would it be possible to write a small example ?

Thanks a lot

sdmueller, 2024-01-05:
Hi @domotiquefacile 

What weird outputs are you getting? Are you able to start measurements with the sensor?

Also did you check if you get an error back from the scd4x calls? (see https://github.com/Sensirion/arduino-i2c-scd4x/blob/master/examples/exampleUsage/exampleUsage.ino for how to read it out)

domotiquefacile, 2024-01-05:
I always got the same magic value 42405, similar to that person here:
https://github.com/Sensirion/arduino-i2c-scd4x/issues/30#issuecomment-1878922692

Same value for getAutomaticSelfCalibration or 
getAutomaticSelfCalibrationInitialPeriod or getAutomaticSelfCalibrationStandardPeriod.

Yes I successfully managed to set the value without error but wanted to confirm it by displaying it.

Can you perhaps write an example to display those values?

Thanks

sdmueller, 2024-01-09:
Your code looks fine and should work.

What I would do is additionally read out the error values. Something like this:
```
    uint16_t ascEnabled;
    uint16_t error;
    char errorMessage[256];
    error = scd4x.getAutomaticSelfCalibration(ascEnabled);
    if (error) {
      Serial.print("Error while trying to execute getAutomaticSelfCalibration(): ");
      errorToString(error, errorMessage, 256);
      Serial.println(errorMessage);      
    } else {
      Serial.print("ASC enabled: ");
      Serial.println(ascEnabled);
    }
```
Based on your return value of 42405 it might be that you are trying to read out the values while a measurement is running.
If you check the data sheet (https://sensirion.com/media/documents/48C4B7FB/64C134E7/Sensirion_SCD4x_Datasheet.pdf) you see in chapter 3.4 that you can't use these two commands during a measurement.
If you use the code above you would then see a corresponding error.

Can you check and confirm that you are executing the commands only when no measurement is running?

domotiquefacile, 2024-01-12:
Oh thank you very much, that was indeed my mistake: I didn't notice that it will not work during measurements.
Now that I execute it outside measurements, it works fine.
Thanks again

closed 2024-01-12 by domotiquefacile