Sensirion/arduino-i2c-scd4x #31: Will there be an example for connecting multiple sensors?

lemorage, 2024-04-09:
The example usage code is quite great for newbies, however, after I connected more than one sensor to the Arduino board, my program crashed. And the provided error log is "Wrong CRC Found", which makes me lose my head. I wonder if there will be some example code to show how to interact with multiple sensors?

Love4yzp, 2024-04-10:
Hi, have you checked the pins? they should not be in conflict.

abrauchli, 2024-04-10:
Is your question with multiple scd4x on the same bus or other sensors?
I²c only allows one same address per bus. Scd41 is 0x62. If you have two devices replying to a query, CRC issues are expected unless they both reply the exact same answer.
I didn't see how to reconfigure the address but you can use another bus if available or otherwise use a multiplexer device.

lemorage, 2024-04-10:
Yes, exactly. I have more than one sensor on the same bus, and scd41 is 0x62 by default. They get into conflicts when I try to run the example code. I know we could change the I2C address using a multiplexer, but can we also change the address in our program?

qfisch, 2024-04-10:
Hello @lemorage ,
You are not the first person to request an example using several sensors. 
We unfortunately did not find the tie yet to do it. I will try to push it out of the backlog.

The general rule for using several of the same sensor:
1. Check if sensor supports several i2C addresses.
2. If not, your board might have more than one i2C bus.
3. If not, you need to use a multiplexer which will distribute the messages across the sensors.

I hope it helps
Quentin

lemorage, 2024-04-11:
@qfisch Cool, I'll try to see if these options solve the problem, anyway it seems this must be something related to the hardware stuff, not something we can do through programming.

Look forward to any updates on the progress.

closed 2024-05-06 by qfisch