Sensirion/arduino-i2c-scd4x #32: What does it mean for an error code returned?

lemorage, 2024-04-11:
I connect the SCD41 sensor and Arduino Uno board with the breadboard and run the example program from this library. The program fails. After some time of debugging, I think the problem is my `scd4x.getDataReadyFlag(isDataReady)` always returns 267, which is a very odd number. I haven't found any useful information explaining the error codes, and the comments for this function just say "@return 0 on success, an error code otherwise".

I wonder if the returned error code has some special meaning depending on the number, just like HTTP response status codes. Or the error code itself doesn't make any sense, which just simply means there is an error somewhere.

sdmueller, 2024-04-23:
Hi @lemorage 

The error codes are platform dependent. I2C in general only knows success or fail. However, our Arduino drivers use the `arduino-core` library which has its own error codes defined in [SensirionErrors.h](https://github.com/Sensirion/arduino-core/blob/main/src/SensirionErrors.h).

If you use the example program as is, the error code will be translated via the `errorToString` function and you should see the resulting error also as text.

Where the error seems to come from in your case:
`scd4x.getDataReadyFlag` uses `SensirionI2CCommunication::sendFrame` from arduino-core, which has some error handling. 267 is the bitwise OR combination of the high level error `WriteError` and the low level error `I2cAddressNack`. This means that there was no response on the I2C address you used. But `errorToString` should translate that already for you.

Hope this helps!

lemorage, 2024-04-24:
Ohh, thank you for your detailed explanation! I suspected there should be something wrong with the I2C devices, but I couldn't find the exact reason. The program just didn't print anything when `errorToString(error, errormsg, 256);` was there, and I had to manually print the return value of `scd4x.getDataReadyFlag` to see if it was 0. So I really thought hard to figure out what is the sense of this error number.

closed 2024-04-24 by lemorage