Sensirion/arduino-i2c-scd4x #33: How to find the implementation of the `errorToString` function, where I think bugs are in it

lemorage, 2024-04-13:
First of all, I try to find the implementation details of the function `errorToString`, which is used in the example usage program. But I can't find it. I think there is something wrong with this function, and the reason is as follows...

I connected a Micro SD Card Adapter (which has an SD card in it) and an SCD41 sensor with the Arduino Uno board, and I confirmed all the wires were correctly connected. I tried to run the program below but failed. Only after I commented out the two lines `errorToString(error, errorMsg, 256);`, the program ran smoothly. I was shocked, but I couldn't find the reason. I hope someone who has the same devices can try the program below, and I wonder if you will find the same error as me. I am very free to talk with you about this very odd bug, which has confused me for three days.

```cpp
#include <Arduino.h>
#include <SensirionI2CScd4x.h>
#include <Wire.h>
#include <SPI.h>
#include <SD.h>

SensirionI2CScd4x scd4x;
File dataFile;

void setup() {
    Serial.begin(115200);
    while (!Serial) {
        delay(100);
    }

    Serial.println("Initializing SD card...");
    if (!SD.begin(4)) {
        Serial.println("SD card initialization failed!");
        while (1) {};
    }
    Serial.println("done.");

    dataFile = SD.open("c_data.csv", O_WRITE | O_APPEND);
    if (dataFile) dataFile.println("CO2, Temperature, Humidity");

    Wire.begin();
    scd4x.begin(Wire);

    uint16_t error; char errorMsg[256];

    error = scd4x.stopPeriodicMeasurement();
    if (error) {
        Serial.print("Error trying to execute stopPeriodicMeasurement(): ");
        errorToString(error, errorMsg, 256); // comment out this line
        Serial.println(errorMsg);
    }

    error = scd4x.startPeriodicMeasurement();
    
    if (error) {
        Serial.print("Error trying to execute startPeriodicMeasurement().");
        errorToString(error, errorMsg, 256);  // comment out this line
        Serial.println(errorMsg);
    }

    Serial.println("Data collection started...");
}

void loop() {}
```

qfisch, 2024-04-17:
errorToString is in arduino-core
see [SensirionErrors.h](https://github.com/Sensirion/arduino-core/blob/main/src/SensirionErrors.h)

closed 2024-04-17 by qfisch

lemorage, 2024-04-17:
Thanks for the link @qfisch 

Can u also share some ideas on why this odd error will happen when the `errorToString` function is not commented out? I am completely using the same code from the example usage.

qfisch, 2024-04-17:
1. Are you running the example without modification or the code you provided above ?
2. What is the error message ?

lemorage, 2024-04-17:
Yes, exactly the same code as I showed above. And actually what I've done was just combine the example code of both (from SD module and SensirionI2CScd4x module).

The weirdest thing is I cannot observe any error msg from the serial monitor. The program just didn't print anything, which looked like it was stuck. Sometimes, it kept printing `Initializing SD card...` again and again. After removing the `errorToString` lines, it prints everything normally (all the data from the sensor). But I just can't see there is anything wrong with the call of `errorToString` function here. I'm kind of stuck.

qfisch, 2024-04-17:
Can you test with the stock example from this repo ? without anything related to your sd card.

lemorage, 2024-04-18:
Are u referring to this [example](https://github.com/Sensirion/arduino-i2c-scd4x/blob/master/examples/exampleUsage/exampleUsage.ino)? I've tried the example from this repo, without anything related to sd card, and it works perfectly. Also, I've tried the example from the sd card repo, and it goes very well.

qfisch, 2024-04-18:
Okay. it could be that your board is running out of memory.
You could try reducing the size allocated to the error message `char errorMsg[16];`

lemorage, 2024-04-18:
Sure. I'll try that later after my board is idle. Thank you for your patience and instructions.