Sensirion/arduino-i2c-scd4x #35: setSensorAltitude below sea level

GlebDen, 2024-06-25:
Hello. 
Is it possible to setSensorAltitude below sea level?

abrauchli, 2024-06-25:
Hi @GlebDen 

Use the fact that the pressure increases by about 1 hPa for every 10 meters you go below sea level along with `setAmbientPressure()` instead of setSensorAltitude(). Note that you can't use both functions together (whichever is called last is used).

Starting from sea level (1013.25 hPa):
For every 10 meters, the pressure increases by approximately 1 hPa.
At the Dead Sea shore in Israel, the lowest dry land point on earth at 430 meters below sea level, the increase in pressure would be 430÷10=43 hPa or approximately: 1013.25 hPa+43 hPa=1056.25 hPa

So `setAmbientPressure(1056);` followed by enjoying some floating in the warm water.

psachs, 2024-07-23:
Thanks @abrauchli for the detailed answer. I will close the ticket as everything should be clear now.

closed 2024-07-23 by psachs