Sensirion/arduino-i2c-scd4x #39: Automatic self-calibration clarification

stas-sl, 2024-09-25:
Hi,

I’m trying to better understand how ASC (Automatic Self-Calibration) works, and which conditions need to bet met for optimal results, as I'm a bit confused. Here’s an excerpt from the spec:

> By default, the ASC algorithm assumes that the sensor is exposed to outdoor fresh air at 400 ppm CO2 concentration at least once per week of accumulated operation using one of the following measurement modes for at least 4 hours without interruption at a time

Could you help clarify a few things?

- Why does ASC need 4 continuous hours of exposure to fresh air for calibration, while FRC (Forced Recalibration) only requires 3 minutes? Why isn’t a shorter exposure to low CO2 levels sufficient for ASC?

- Can the ASC calibration period be extended to a month or longer, or is it necessary to recalibrate every 7 days?  I mean, could it really loose accuracy in just a week?

- Does ASC keep a history of CO2 measurements throughout the entire calibration period (meaning longer periods would require more memory and could risk overflow), or does it simply record the lowest detected value for the whole period?

- If my sensor operates only 6-8 hours a day, will ASC still recalibrate every 7 calendar days, or does it only count active hours? In that case, would it wait until the total runtime reaches 168 hours (7 days of operation), which could take much longer than 7 calendar days?

Thanks!

abrauchli, 2024-09-25:
Hey @stas-sl these are proprietary algorithms that the manufacturers don't give out so the documentation and reverse engineering is as good as it gets.

In terms of why the difference between FRC and ASC, FRC has to be run in a known environment and will immediately readjust the internal baseline. Think of it as a relative sensor which only outputs absolute values with a good baseline value. Unfortunately those baselines drift (well, actually it's the sensor/materials etc that drift but it's relative so it doesn't matter which), so you have to run an algorithm that tries to guess the environment. Since you can't go below ~400ppm CO2, the lowest read value is likely to be assumed as 400.. But outside of FRC you don't know if the lowest value is actually 400 or just relatively lower to whatever you had before, so an adjustment will take place slowly. It likely doesn't matter if it's exactly 7 days but that's what the manufacturer chose in order to guarantee the specified tolerances.
The more often and longer you can expose it to the low CO2 value the better. Think of it as walking blindly in a straight direction. Every time you open your eyes you can readjust your direction and the longer you go with your eyes closed, the more off you'll be.
Hope this helps

stas-sl, 2024-09-25:
Thanks for quick reply, though it doesn’t help much unfortunately. It all makes sense, but the reason why I’m asking all these questions is because, I realize that it probably wouldn’t be the case when the sensor is exposed to fresh air for long periods of time, as it is powered on only when there is somebody in the place, but when people leave, the whole place is left without electricity, though even if there was electricity, all windows and doors are closed, so there is no guarantee that CO2 level will drop to 400 ppm at night.

 You say: 

> The more often and longer you can expose it to the low CO2 value the better

but my goal is quite opposite in a sense: I’d like to find minimum required exposure to low levels of CO2, that would be sufficient to keep it accurate.  If ASC really requires 4 continuous hours of operation in fresh air each week to function properly, I'm concerned it might cause more harm than good in my case. The risk of incorrectly setting the baseline to 600 ppm seems greater than the potential drift by a few percent over several months.

I still don't quite understand how exposing the sensor to fresh air for longer periods helps the ASC algorithm determine that it's truly 400 ppm, and not 600 ppm or another concentration. I'm definitely not an expert and haven't spent much time on this, but right now, I can't think of a better approach than the simplest one: just tracking the lowest value over a certain period and assuming that's 400 ppm, regardless of how long it stayed at that level. 

I’m just trying to think logically, as in the absence of more details about algorithm that’s all I can do. Let’s consider a few scenarios:

1. A sensor is never exposed to fresh air over a week. The lowest concentration was 600 ppm and it stayed at that level for more than 4 hours.
2. Same as 1, but sensor was exposed to fresh air for short periods of time let’s say 5-10 minutes a few times a week, but the remaining time it was higher than 400 ppm.
3. Sensor is exposed to fresh air (~400 ppm) regularly for long periods of time.

Case 3 is the simplest and perfect case, ASC will work the best and there is nothing to discuss here. 

Case 1 seems fairly straightforward as well - since no levels below 600 ppm were detected, the algorithm has no choice but to assume it's fresh air and set the baseline accordingly (and wrongly).

Now for the most challenging scenario - Scenario 2. It clearly doesn’t meet the requirement of 4 uninterrupted hours of fresh air. So, what would happen here? On one hand, there's not enough exposure, but on the other, there is still some. Will it ignore those shorter periods and set the baseline at 600 ppm, since that lasted much longer? Or will it still take the brief exposure into account and calibrate properly at the lowest levels? I’m struggling to understand why it would be worse to account for the short periods of exposure, even if they weren’t long enough.

abrauchli, 2024-09-25:
The baseline is likely set in a logarithmic way: when the values are much lower than your regular environment (ventilation event) your baseline would drop much quicker at the beginning, so even if you have just a few minutes of exposure you'd be approaching the calibrated value faster. Even a reading of 600ppm is still considered excellent air.

In North America, buildings are less well insulated than in Central Europe - esp. buildings with forced air heating leave enough room under the door for return air so your CO2 will dissipate quite fast.. if a room doesn't have outside-facing windows, leaving the meeting room door open between meetings is usually the best you can do CO2 wise anyways.

For testing purposes: if you have portable devices, you can try moving them to a well-ventilated spot (beneath a window or an electric outlet on an outwards facing wall) every couple of weeks and see if that changes the readings. I would assume you wouldn't see much of a difference. Otherwise you can try the SCD30 (or any SCD3x). Those work with NDIR and might be a little less sensitive to your use case but they still self-calibrate to 400.
https://sensirion.com/products/catalog/SCD30

I'd also be delighted to consult professionally on the software side.