Sensirion/arduino-i2c-scd4x #47: SCD41 error: invalid sample detected

Jelle7and9, 2025-01-31:
Afther a couple of months I have this problem.  Is same as issue #15 I think.
Co2 measuring results in **invalid sample detected, skipping**.
A selfTest result to code **196** (sensorstatus). CO2 readings give 0, but temperature and humidity do work.
My sensormodule is GY-SCD41 Module with ESP32S3 board, but also tested on Arduino UNO.
Tried 3 different boards and several example codes. All the same outcome.
Also tried with 5V external power, makes no difference.

Please advice if there is a solution.
Or is this meaning the sensor is faulty?

[image]

LeonieFierz, 2025-02-03:
Hi @Jelle7and9 

Thanks for contacting us. 

Self Test Code 196 indicates several errors on the sensor. This could be due to ASC conflicting when the sensor read out is triggered without checking the data ready flag. 

To recover, please try to issue a performFactoryReset command and power cycle the sensor. Then, make sure to poll the getDataReadyStatus before issuing a readMeasurement command.

Let me know if this resolved your problem.
Best, Léonie

Jelle7and9, 2025-02-04:
Did not resolved this problem unfortunately.

LeonieFierz, 2025-02-05:
@Jelle7and9 I will ask back to our product group if they can give more details what could be the cause for the self test status 196 and come back to you as soon as I have more information.

Jelle7and9, 2025-02-06:
Details:

Board: ESP32S3 (BPI Leaf S3)
SDA pin 15 / SCL pin 16 / 5V pin to sensor (also 3.3V tested)
SDA & SCL with pullup resistor 2K, 4.7K and 10K tested
_Factory reset sketch executed_

Arduino IDE v2.3.4
esp32 by Espressif v3.1.1
SensirionI2cScd4x v1.0.0
Windows 11 23H2

Basic I2C scanner:  I2C device found at 0x62 
Sensirion SCD4x exampleUsage sketch:
_serial number: 0x5..........3
CO2 concentration [ppm]: 0
Temperature [°C]: 21.10
Relative Humidity [RH]: 36.36_

Also used this sketch for testing with different speeds.
[https://github.com/RobTillaart/MultiSpeedI2CScanner](url)
outcome:

[image]

LeonieFierz, 2025-02-13:
@Jelle7and9 

I talked back to the product group and they tell me that the error 196 indicates that something is odd with the power supply of the sensor. Please make sure the power connection is good and enough power is supplied, as the sensor uses up to 200mA while measuring CO2.

If this still does not resolve the issue please contact us on info@sensirion.com for further investigations.

Best, Léonie

Jelle7and9, 2025-02-18:
Dear Léonie,

I ordered an exact same sensor and this is working as suposed to be.
I exchanged the picoblade cable and the faulty sensor did not worked, again no CO2 value. Only temp. and hum.
A continuity test with a multimeter shows good contact between socket pins and the sensor pins.

LeonieFierz, 2025-02-19:
Dear Jelle7and9

Thank you for the feedback. 
Hearing that, it looks like the sensor itself has a problem. If you want to follow up, please contact us on info@sensirion.com to probably arrange to send it back so that we can analyze further.

Best, Léonie

closed 2025-02-19 by LeonieFierz