Sensirion/arduino-i2c-scd4x #51: getSensorVariant не работает.

SoftFelix72, 2025-05-06:
getSensorVariant не работает, всегда обнаружен SCD40.

SensirionI2cScd4x.cpp line 174.

uint16_t variant = (uint16_t)(rawSensorVariant & 4);

Надо проверять bit 12 (0x1000), а проверяется bit2 (& 4, 0x0004)

variant всегда равен 0.

Далее проверку тоже надо изменить.

wollewald, 2025-05-30:
Agree with @SoftFelix72. But not everyone may understand Russian language. So, maybe this helps:

Line 174 should be changed to:

uint16_t variant = (uint16_t)((rawSensorVariant & 0x1000)>>12);

SoftFelix72, 2025-05-31:
> Agree with [@SoftFelix72](https://github.com/SoftFelix72). But not everyone may understand Russian language. So, maybe this helps:
> 
> Line 174 should be changed to:
> 
> uint16_t variant = (uint16_t)((rawSensorVariant & 0x1000)>>12);

Hi. Yes, now it's correct and it will work. I didn't shift by 12, but compared directly with 0x0000 and 0x1000. :) So your code is more beautiful. In any case, the library needs to be officially fixed.

qfisch, 2025-06-30:
Fixed in [PR#54](https://github.com/Sensirion/arduino-i2c-scd4x/pull/54)

closed 2025-06-30 by qfisch